espressif/esp32-wifi-lib #1: Source code?

danielinux, 2016-09-06:
The LICENSE file in this repository refers to APL 2.0, but no sources are available. 

Could you please release the source code of the library as specified by the license terms?

projectgus, 2016-09-06:
The Apache License grants anyone the right to redistribute the software in either source code or object code forms, under the terms of the license. It doesn't require anyone to provide source code to anyone else.

So you're welcome to both use and redistribute these libraries of object code under the terms of the license, but as Espressif doesn't provide the source code for these libraries there's no way to use or redistribute any source code (even though it is permitted by the license.)

The Apache License is used here (even though the libraries are in binary form) because it's a well understood license. It also matches with the license on the source code in the parent, open source, [esp-idf repository](https://github.com/espressif/esp-idf.git). This means you only have to put one set of license notices on a redistribution that combines these projects.

Everyone here agrees that source code is useful and open source is awesome. We want to open source as much as we can in esp-idf. The low level Bluetooth & WiFi implementations are proprietary and will continue to be provided as binary libraries, but all other software is either available as source code under a recognised open source license, or we plan for it to be.

There are also plans to make integrating these libraries into other environments straightforward, although for now they only work with esp-idf.

closed 2016-09-06 by projectgus

danielinux, 2016-09-08:
Weird, for APL is an Open Source license. One could even think you are abusing the "Open Source" terminology for marketing purposes (especially after reading your comment about the "well-understood license"). Perhaps not infringing the license, but still very awkward. Publishing a whole binary-only repository with an "Open-Source" logo on it.

ESP32DE, 2016-09-08:
not sure what the helpfull point of yours is, but 
 | Publishing a whole binary-only repository with an "Open-Source" logo on it.

not sure again, that you have seen the "Open-Source" main and the split to the lib that you spotted,
there is linked out to the spotted repo from the main: 

https://github.com/espressif/esp-idf/tree/master/components/esp32

you must click example on the lib link:
 https://github.com/espressif/esp32-wifi-lib/tree/b95a359c344cd052f45e3c38515e4613dd29f298

then you come to the part that you spotted as "a whole binary-only repository".

i see an esp-idf with a linked WiFi ( next time BLE too ) lib repo 
you cant use the spotted repo allone - you must use the esp-idf with the spotted repo.
not sure again, what you want to do with the libs only  ;-) or why the excitement.
ah.. see - closed so i think it was clear.

projectgus, 2016-09-08:
> One could even think you are abusing the "Open Source" terminology for marketing purposes 

The words "open source" don't appear in the README or the license text, or anywhere in the git repository.

However, to help alleviate any misunderstanding I've updated the README to make it clearer what it is in this repository (and why it is licensed the way that it is).

> > The low level Bluetooth & WiFi implementations are proprietary
> Well, they aren't if you are distributing them with the Apache license.

The Apache License is approved by the Open Source Initiative as a valid open source license, meaning that source code licensed under it is considered open source. This doesn't mean that everything the license is applied to is automatically open source.

Quoting from the Definitions section of the license text:
> "Object" form shall mean any form resulting from mechanical transformation or translation of a Source form, including but not limited to compiled object code,
> ...
> "Work" shall mean the work of authorship, whether in Source or Object form, made available under the License,

This Work is in Object form.

Perhaps you are confusing the Apache License with a strong copyleft license such as the GPL?

>  "well-understood license"

I mean well-understood legally. We need to provide some kind of distribution license with these libraries, so people are granted rights to use them and re-distribute them. Providing everything (esp-idf & these libraries) under one license reduces the legal complexity, compared to mixing in our own custom license.

danielinux, 2017-01-10:
IANAL, but the Open Source Initiative lists apache license among popular license, and the contradiction is if you check the first paragraph here: https://opensource.org/licenses

Anyway, it seems that you are heading in the right direction by considering #2 - A bunch of potential users is looking forward to that. And there are technical reasons, for one the possibility to integrate different open source solutions for connectivity at TCP/IP level than the lwIP you propose.

igrr, 2017-01-10:
These libraries do not provide TCP/IP layer — it is in the open source part (ESP-IDF).

There is a [tcpip_adapter](https://github.com/espressif/esp-idf/tree/master/components/tcpip_adapter) component in ESP-IDF which acts as a facade for the TCP/IP stack, allowing customers to use different stacks, if needed.

danielinux, 2017-01-10:
@igrr this is interesting, but could you please point me to datalink send/receive facilities, overriding lwIP networking layer?

igrr, 2017-01-10:
For WiFi, these are `esp_wifi_internal_tx` and `esp_wifi_internal_reg_rxcb` APIs ([link](https://github.com/espressif/esp-idf/blob/master/components/esp32/include/esp_wifi_internal.h#L72)), which are provided by binary libraries in this repo.
IP stack can use these APIs to send and receive data frames to/from the lower layer. 

For Ethernet, these are `esp_eth_tx` and `eth_tcpip_input_func` (`eth_config_t` member), provided by the [ethernet component](https://github.com/espressif/esp-idf/tree/master/components/ethernet).

danielinux, 2017-01-10:
Awesome, thanks.
So, this means I can provide a GPL firmware to run on my ESP32, and consider the library as a separate component, released under a permissive license, following this: http://www.softwarefreedom.org/resources/2007/gpl-non-gpl-collaboration.html ?

How does this not clash with the fact that part of my firmware (as a derivative) does not have all the sources available?

igrr, 2017-01-10:
Although the article you link talks about "permissively licensed files" in general, it is implied that those files are *source* files (as opposed to object files). Obviously if you plunk a set of closed source compiled libraries into your project and release it under GPL, you will have trouble providing the source code of the closed source part to the downstream users. There are ways of working around that (such as dynamic loading) which I'm not going to detail into.

As @projectgus has mentioned above, in this case the Apache license doesn't indicate that this is an open source component (even though Apache license *is* recognized as an open source license). This license indicates that we don't restrict usage and distribution of these libraries (so it is okay to include them into any project as long as the project license allows that).
It is convenient to distribute the main project (ESP-IDF) and the libraries under a single license, that's why Apache license is used here.

danielinux, 2017-01-10:
OK, this is very clear now. Thanks a lot for the clarification. Indeed this is also explained in https://opensource.org/faq#non-distribution 

I will then wait until all the source code is released before starting to use ESP32.