espressif/esp32-wifi-lib #4: Need additional station connect parameters

negativekelvin, 2017-02-28:
It takes 1s longer to boot and connect to an ap on channel 11 than on channel 1.  After the first time we connect to an ap, we can remember its channel.

So we need to be able to provide a prioritized list of channels to use when scanning in sta mode.  Also we should be able to specify channel dwell times.  It is assumed that directed probe request is used here.

Also related: https://github.com/espressif/esp-idf/issues/326

negativekelvin, 2017-04-17:
@TimXia any info on this topic?

TimXia, 2017-04-18:
@negativekelvin I guess that you mean fast connect. We will try to implement what you say.

TimXia, 2017-05-04:
@negativekelvin This feature will be implemented in IDFv2.1.

negativekelvin, 2017-05-05:
@TimXia Thanks for adding these features.  

Regarding  

> If AP is not on the specified channel, continue scanning the next
> channel

I think there is a case when we don't want to do that, APSTA mode so softAP can still operate on the channel.

TimXia, 2017-05-05:
@negativekelvin The channel of softAP follows that of STA. So, I think it does not impact softAP too much if STA continues scanning the next channel.

negativekelvin, 2017-05-05:
@TimXia yes but if there are client connected to softap we don't want to change channel on them.  Also if we know for certain AP channel we may not want to use battery to scan other channel.

negativekelvin, 2017-05-06:
@TimXia also if we look at some [data](https://wigle.net/stats#octetstats) we see that most ap on channel 1,6,11 so we want to be able to scan these channels first before other channels.