espressif/esp32-wifi-lib #6: What does WIFI_STORAGE_FLASH do?

edmund-huber, 2017-03-19:
What exactly does WIFI_STORAGE_FLASH store to flash memory? And how is it reused?

This is the only documentation I could find about that flag in esp-idf: `NVS is used to store WiFi data if the :doc:esp_wifi_set_storage(WIFI_STORAGE_FLASH) <api/wifi/esp_wifi> initialisation function is used.`

If I can't figure out what exactly it stores (and how it stores it) and when it reuses the info, I don't feel comfortable using WIFI_STORAGE_FLASH.. thanks in advance.

shirish47, 2017-06-29:
did you get to know how it works or what it does?

mrmonteith, 2018-06-13:
This adds some tidbit of information:
<https://esp32.com/viewtopic.php?f=13&t=316>
  Several of their examples include a flash init but then don't seem to save information.   A couple others store IP Address and some store other connection information.  You can store the information in RAM or NVS(non-volitile-storage). 

  I'm guessing the main reason is recovery.  For instance you go into a sleep mode.  Instead re-establishing a new connection after you come out of sleep mode you restore the previous connection by retrieving the values from storage. 

 However pretty all I've seen is writing to it but unless I missed something didn't see them pulling back information.  Some of the other just initialize the flash but don't save or read from it.   So I think for the most part you can comment out the include:
`#include "nvs_flash.h"`
and any lines that makes calls to initializing the flash. 

 What I don't know is what non-volatile storage is?  I don't know if it's eeprom or not.  But mainly wherever it's stored is it limited to a number of read/writes?  If it is limited then we have to make sure we don't mess with it no more than necessary.

cyberman54, 2018-12-09:
@mrmonteith Since ESP32 has, according to datasheet, no embedded NVS i assume that the Wifi config is written to external memory, connected by SPI to the chip. So it depends on the type of external memory which is used how many flash cycles we have.

https://www.esp32.com/viewtopic.php?t=709

q7777777hk, 2020-02-06:
Why esp_wifi_set_config() sometimes resulted in ESP_ERR_NVS_NOT_ENOUGH_SPACE ?
1) I saw my NVS usage is only 
Count: UsedEntries = (198), FreeEntries = (306), AllEntries = (504)
This sometimes happens not always
2) I understand 1st time all esp_wifi_set_config() may require memory, but this error happen from time to time, seems to me it keep request new NVS memory for each call

sukesh-ak, 2022-11-16:
Here is the relevant header with explanation 

https://github.com/espressif/esp-idf/blob/056d72ece5e073fbe5b55d9468c5c408b1c189f3/components/esp_wifi/include/esp_wifi_types.h#L327