espressif/esp32-wifi-lib #17: Document the SDR that this code internally uses?

wgaylord, 2019-01-18:
Based of the technical docs of the core https://www.espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf#page24

It be interesting to have access to that SDR directly. For experimental usage as a ham radio operator.

wgaylord, 2019-03-27:
Any comment on this?

igrr, 2019-03-27:
Could you please clarify what is the "SDR" that you reference in the datasheet? If you mean "software defined radio", then I don't think there are such capabilities on the chip. Wi-Fi radio peripheral is not documented in the TRM, so there is nothing that can be done in ESP-IDF documentation to resolve that, unfortunately. You can raise TRM related requests in https://esp32.com/viewtopic.php?f=5&t=905.

wgaylord, 2019-03-27:
On page 24 of the Technical Docs (3.4.1 and 3.4.2) it does say that wifi and Bluetooth are based around a set of high speed DAC and ADC. (Specifically in quadrature baseband converted to digital which is exactly how an SDR works.)

Are you trying to tell me that the WIFI stack that implements that whole interface to those can't be documented any further then the vague information in the technical documentation?

wgaylord, 2019-03-29:
Seems weirdly the docs mention and describe these DACs and ADCs that you can't even get access to. :(

closed 2019-03-29 by wgaylord