espressif/esp32-wifi-lib #22: Passive doppler radar for motion detection using ESP32 wifi HW

Harvie, 2019-10-12:
Hello,
maybe you are familiar with doppler "radar" modules for detecting motion using RF energy reflected by human body back to detector. Like the famous RCWL-0516 (using ~3.2GHz band):

[image]

I am quite sure that something very similar can be achieved with 2.4GHz frontend of ESP32 chip. Eg.: transmitting some short RF signal (eg. esp-now packets) and measuring the reflected frequency for doppler shift.

That way you can detect if there is some motion near the ESP32 chip (and possibly measuring distance of the moving human/object to artificialy limit the range of motion detection).

https://www.researchgate.net/publication/283649280_A_real-time_high_resolution_passive_WiFi_Doppler-radar_and_its_applications

Do you thing there are chances that espressif will look into this topic? I believe that espressif engineers would do best job implementing such feature, since they already have knowledge of inner workings of the RF frontend hardware.

nuschpl, 2023-12-29:
While we can probably agree open solutions are better than DRMs and propretory code,  devices RF frontends are closed and certified for a reason - the RF specturm has dedicated part for communication which is WiFi and dedicated for the radars. So while there are whitepapers which already made concept you described real and it's good for allowing the scienice go forward IMHO its not good idea to such feature generally available for script kiddies on general purpose Wifi/BT chip. As the cost of your joy would be your neighbour not able to realiably connect VPN .  I doubt Espressif would involve having in mind RF device certification process

dgoss01, 2024-10-29:
This can and has been achieved, and no, it does not affect the Wi-Fi signal.  The ESP32 is not necessarily high resolution, but it can detect motion through wi-fi.  Using a good quality WiFi card and antennas on a PC, researchers have been able to detect location and body position of people inside a building. 
https://github.com/paoloinverse/bistatic_interference_radar_esp

Huzzla101, 2024-10-31:
Hi thanks so much for your responce and interest. Let me just tell you that
ive got mine and it results is way over my expectation , this is like a
gold mine for people who understand... its amasing... i can hold my hand to
a wall and it scans arms and give out about 32-38 degress depends on my
mode, but wall itself is cold and showing a frequent warmth all over wall
but not pointing out anything besides my arm that is glowing .... BUT if i
take the hand away it starts to scan heat from other side of wall instead
so i can actually see..... alot more then expected.. Wow amasing... And its
open source to for later upgrades... But its already "there"... Simply
amasing.. Im gonna get me couple more of them as i can add them to my
router and cloud and use it even as a motion sensor detector with alarm
etc. It can also peak out to cold areas to ..if ur neighbour have below
20degress flowing thru his walls then go knock  on his door .p  but it
should be around 23-35 celsius in any house ... Thanks everyone for yourt
great efforts and collobration...

Huzzla101, 2024-10-31:
BTW... I live in  Scandinavia north Europe so here its starting to get
winter and soon minus degrees so it depens where u live off course hehe

Harvie, 2024-10-31:
> but wall itself is cold and showing a frequent warmth all over wall

wait what? this is not about an IR camera and i don't think temperature should be a factor here...

Huzzla101, 2024-10-31:
No , wall was cold yes below 20C  so when i took away my hand it scanned
the temprature coming from behind the wall i could easily see what part of
the wall that had most heat and it started to form figures of the heat
arriving, as i could easily see where the heat warmer was etc , but the
link given here was amasing that was a totally different link about radar
scanning , exactly what i was looking for. so all i need now is some good
programmer who can integrate this radar scanning into the termal heat
sensor camera i have that allready have it almost integrated but off course
if we can update it a little better then im in... Future plans hehe....

Huzzla101, 2024-10-31:
Here is overview... You are correct that myself uses and talking about a
termal radar HEAT sensor camera that senses heat from 20C to 400C and its
already a very goo dprgramming behind uit and i can allready see warmth
thru houses but i was asking if anyone had some radar scanner ideas like
the ones they use at hospitals etc , as military is using that radar tech
together with termal heat sensors when they scan areas and houses for heat
, one let them see thru all walls and one let them see any heat. i can do
almost all that but i want a better radar integrated for seeing thru houses
etc not just from a heat sensor cam but other cams , in end will be a good
program for radar scanning from normal cams  from ip cams to mobile etc...
its just useless apps now.. its a multi million idea to if u good at making
APK insstall files ec.... and its free to use , if u are a pogrammer that
is