espressif/esp32-wifi-lib #31: About increasing the number of WiFi MAX STA connections. 关于增加WiFi的最大STA连接数

FPOOOOOOO, 2022-07-18:
I am using the code in "esp-idf/examples/wifi/getting_started/softAP", with version ESP-IDF 4.4 dirty.  Here is my step:
First: I change the value "Maximal STA connections" in "idf.py" to 16  
Second: change the value "ESP_WIFI_MAX_CONN_NUM" in "esp-idf/components/esp_wifi/include/esp_wifi_types.h" to 16 as well
Then I flash into ESP32_DevKit , then use STATION to connect the AP. 
When the 11st STA join , the log print "wifi:max connection, deauth!" it seems that it can't be more than 10. But I really need to use 16 station STRAIGHT connect to the AP, not by AP-STA mode or Mesh. Please help to solve the limit of number of connection. If possible, please offer me a version of " libnet80211.a" that the ESP_WIFI_MAX_CONN_NUM is 16.

您好，我正在用 "esp-idf/examples/wifi/getting_started/softAP" 的代码，版本是ESP- IDF 4.4。
1、首先我更改了idf.py里的Maximal STA connections到16
2、然后我修改了esp_wifi_types.h 里的ESP_WIFI_MAX_CONN_NUM 到16
然后烧录到板子里，第11个STA连上AP的时候，得到了wifi:max connection, deauth！ 说明这样子增加没有用。我真的很需要1主16从的直连方式，而不是通过AP STA或者Mesh模式转接。请尽快帮助我解决这个限制。
如果可以的话能够给我一个 libnet80211静态库，里面的ESP_WIFI_MAX_CONN_NUM限制是16，这样我可以用到项目里。

FPOOOOOOO, 2022-07-19:
Here is my log:

> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6624
load:0x40078000,len:14780
load:0x40080400,len:3792
entry 0x40080694
[0;32mI (27) boot: ESP-IDF v4.4-dirty 2nd stage bootloader[0m
[0;32mI (27) boot: compile time 11:52:46[0m
[0;32mI (27) boot: chip revision: 1[0m
[0;32mI (30) boot_comm: chip revision: 1, min. bootloader chip revision: 0[0m
[0;32mI (37) boot.esp32: SPI Speed      : 40MHz[0m
[0;32mI (42) boot.esp32: SPI Mode       : DIO[0m
[0;32mI (46) boot.esp32: SPI Flash Size : 2MB[0m
[0;32mI (51) boot: Enabling RNG early entropy source...[0m
[0;32mI (56) boot: Partition Table:[0m
[0;32mI (60) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (74) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (82) boot:  2 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (89) boot: End of partition table[0m
[0;32mI (93) boot_comm: chip revision: 1, min. application chip revision: 0[0m
[0;32mI (101) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=13370h ( 78704) map[0m
[0;32mI (138) esp_image: segment 1: paddr=00023398 vaddr=3ffb0000 size=03810h ( 14352) load[0m
[0;32mI (144) esp_image: segment 2: paddr=00026bb0 vaddr=40080000 size=09468h ( 37992) load[0m
[0;32mI (159) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=6d3b0h (447408) map[0m
[0;32mI (322) esp_image: segment 4: paddr=0009d3d8 vaddr=40089468 size=0ae7ch ( 44668) load[0m
[0;32mI (340) esp_image: segment 5: paddr=000a825c vaddr=500000ect name:     wifi_softAP[0m
[0;32mI (387) cpu_start: App verect name:     wifi_softAP[0m
[0;32mI (387) cpu_start: App version:      v4.4-dirty[0m
[0;32mI (392) cpu_start: Compile time:     Jul 18 2022 20:28:44[0m
[0;32mI (398) cpu_start: ELF file SHA256:  2a132ad0197da055...[0m
[0;32mI (404) cpu_start: ESP-IDF:          v4.4-dirty[0m
[0;32mI (409) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (416) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (423) heap_init: At 3FFB74B0 len 00028B50 (162 KiB): DRAM[0m
[0;32mI (429) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (435) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (442) heap_init: At 400942E4 len 0000BD1C (47 KiB): IRAM[0m
[0;32mI (449) spi_flash: detected chip: generic[0m
[0;32mI (452) spi_flash: flash io: dio[0m
[0;33mW (456) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.[0m
[0;32mI (471) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (544) wifi softAP: ESP_WIFI_MODE_AP[0m
I (564) wifi:wifi driver task: 3ffc0624, prio:23, stack:6656, core=0
[0;32mI (564) system_api: Base MAC address is not set[0m
[0;32mI (564) system_api: read default base MAC address from EFUSE[0m
I (584) wifi:wifi firmware version: 7679c42
I (584) wifi:wifi certification version: v7.0
I (584) wifi:config NVS flash: enabled
I (584) wifi:config nano formating: disabled
I (594) wifi:Init data frame dynamic rx buffer num: 32
I (594) wifi:Init management frame dynamic rx buffer num: 32
I (604) wifi:Init management short buffer num: 32
I (604) wifi:Init dynamic tx buffer num: 32
I (604) wifi:Init static rx buffer size: 1600
I (614) wifi:Init static rx buffer num: 10
I (614) wifi:Init dynamic rx buffer num: 32
[0;32mI (624) wifi_init: rx ba win: 6[0m
[0;32mI (624) wifi_init: tcpip mbox: 32[0m
[0;32mI (624) wifi_init: udp mbox: 6[0m
[0;32mI (634) wifi_init: tcp mbox: 6[0m
[0;32mI (634) wifi_init: tcp tx win: 5744[0m
[0;32mI (644) wifi_init: tcp rx win: 5744[0m
[0;32mI (644) wifi_init: tcp mss: 1440[0m
[0;32mI (644) wifi_init: WiFi IRAM OP enabled[0m
[... 44 lines cut ...]
[0;32mI (259184) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.8[0m
I (287224) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
I (287224) wifi:station: 82:a8:75:1c:73:47 join, AID=8, bgn, 40U
[0;32mI (287244) wifi softAP: station 82:a8:75:1c:73:47 join, AID=8, Now has 8 STA[0m
[0;32mI (287414) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.9[0m
I (337154) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
I (337164) wifi:station: aa:67:f3:dc:52:8c join, AID=9, bgn, 20
[0;32mI (337184) wifi softAP: station aa:67:f3:dc:52:8c join, AID=9, Now has 9 STA[0m
I (341404) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
I (341404) wifi:station: 8c:4b:14:14:77:b8 join, AID=10, bgn, 40U
[0;32mI (341434) wifi softAP: station 8c:4b:14:14:77:b8 join, AID=10, Now has 10 STA[0m
[0;32mI (341744) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.11[0m
[0;32mI (342304) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.10[0m
I (350124) wifi:max connection, deauth!
I (351134) wifi:max connection, deauth!
I (352144) wifi:max connection, deauth!
I (353154) wifi:max connection, deauth!
I (356364) wifi:max connection, deauth!
I (357344) wifi:max connection, deauth!
I (358364) wifi:max connection, deauth!
I (359374) wifi:max connection, deauth!
I (360174) wifi:max connection, deauth!
I (360174) wifi:max connection, deauth!
I (361174) wifi:max connection, deauth!
I (362194) wifi:max connection, deauth!

lijunru-hub, 2022-07-19:
非常抱歉，受限于板载资源和稳定性等原因，只能支持最大连接数为10。

FPOOOOOOO, 2022-07-27:
乐鑫研发人员告诉我这个是考虑到稳定性和速率等要求不能随意更改，所以写死在静态库里了

FPOOOOOOO, 2022-07-27:
可以考虑用mesh代替了

closed 2022-07-27 by FPOOOOOOO