espressif/esp32-wifi-lib #42: Use more efficient filter configuration for BSS scan.

Frostie314159, 2025-01-09:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

Hi,

so this maybe a bit unusual, but I'd like to suggest a different hardware packet filter configuration for BSS scanning. I'm a contributor to the [esp32-open-mac](https://esp32-open-mac.be) project, and we recently found the register, which controls scanning mode. This is set in `hal_mac_rx_set_policy`(libpp.a), where the value `0x110` is OR'ed into a register, when RX policy number 3 is set. This is the case, when the upper layers initiate a BSS scan. 

In the past few days, we were able to identify the meaning of the individual bits. Setting bit 4 (`0x10`) allows beacon frames to pass the RX filter in the MAC peripheral, and bit 8 (`0x100`) allows all frames (except control frames) to pass the RX filter.
For a plain BSS scan, setting only bit 4 is sufficient, but currently bit 8 is enabled as well. Simply not setting this bit will reduce the number of frames the CPU has to process drastically, especially on a busy channel. This also does not affect probe requests, since those will still pass the normal MAC address filter, when the RA matches. 

I hope, that we're able to help with this and make the official stack benefit from the research.

Greetings
Simon

Frostie314159, 2025-01-14:
I think I may need to correct myself here a little. Bit 8 doesn't let everything through, but only some frames. I haven't been able to determine the exact behaviour yet.

Huzzla101, 2025-01-17:
Wow thanks for the update , thats very informative, its a very interesting
world we are living in indeed
=)